boneIO-eu/boneIO #61: CAN-BUS

hanusek, 2021-11-15:
Witam.

Na wstepie chciałem pogratulować pomysłu i jego realizacji.  Do szczęścia w waszym projekcie brakuje mi magistrali CAN. 
Czy rozważacie dodanie magistrali CAN? 

https://www.beyondlogic.org/adding-can-to-the-beaglebone-black/

pozdrawiam

maciejkrasuski, 2021-11-15:
Sterownik w gruncie rzeczy jest na to gotowy, bo rs485 jest lutowany na oddzielnej płytce, więc zamiast niego można wrzucić transceiver can i wykorzystać gniazda od rs.

hanusek, 2021-11-16:
> Sterownik w gruncie rzeczy jest na to gotowy, bo rs485 jest lutowany na oddzielnej płytce, więc zamiast niego można wrzucić transceiver can i wykorzystać gniazda od rs.

Hmmm. Nie bardzo... bo sterownik wykorzystać do komunikacji Rs-485 (modbus master) i CAN.

maciejkrasuski, 2021-11-30:
To na to niestety nie mamy już miejsca na płytce ;/

hanusek, 2021-12-01:
> To na to niestety nie mamy już miejsca na płytce ;/

Szkoda. 
DI/DO można łatwo rozbudować przez ekspandery natomiast RS-485/CAN to już gorsza sprawa.

kamil2466, 2021-12-01:
Ja mam pomysł na dołożenie tego CAN tylko musze przeanalizować tą opcje

hanusek, 2021-12-09:
> Ja mam pomysł na dołożenie tego CAN tylko musze przeanalizować tą opcje

Super.

hanusek, 2022-03-31:
> Ja mam pomysł na dołożenie tego CAN tylko musze przeanalizować tą opcje

Udało coś się z tym zrobić?
[Tu](https://www.thomas-wedemeyer.de/de/electronic/arm/beaglebone/canbus-python/) jest spoko tutorial BBB + SN65HVD231.

kamil2466, 2022-04-15:
@hanusek CAN jest dodany do V0.4 input board