boneIO-eu/boneIO #68: [Software/Enhancement] Configuration from SD card with autonomous mode

wstrzalka, 2021-11-22:
It would be nice if there was option to place the static configuration in some readable format (yaml??) on the SD Card. 

That could serve 2 purposes: 
- to define MQTT topics for inputs/outputs on the card instead of sending it from the HA 
- add simple autonomous mode that will hardwire inputs to outputs (for momentary and statefull switches - eg. input 7 toggles output 10; output 11 is directly driven by the state of input 8) - that mode could be either used upon HA failure or even add option to deploy boneIO in standalone mode

pszafer, 2021-11-22:
You're more than welcome to contribute in my PR https://github.com/maciejkrasuski/boneIO/pull/66
My goal is to have on bbb:
- yaml config
- asynchronous, so it won't create so many threads (currently one thread per input)

wstrzalka, 2021-11-22:
I would love to but although I'm a software developer my python skills are close to zero. Happy to help with Java/C/C++ 

At first glance your MR looks like it already does exactly what I had proposed - doesn't it?

maciejkrasuski, 2021-11-30:
I agree - looking at @pszafer yaml schema it looks like he already doing it ;)

pszafer, 2021-11-30:
yes, that's correct

closed 2022-02-25 by maciejkrasuski