boneIO-eu/boneIO #72: Input Board ESP32 v0.1

kamil2466, 2021-12-06:
Witam, stworzyłem schemat płytki dla ESP32 chciałem aby osoby które maja doświadczenie pomogły trochę tworzeniu projektu, może macie jakieś sugestie lub chcecie zmian albo gdzieś jest błąd to proszę o komentowanie.
[input board ESP32.pdf](https://github.com/maciejkrasuski/boneIO/files/7662007/input.board.ESP32.pdf)
[image]

pepsonEL, 2021-12-07:
Kurczę nie za bardzo jeszcze ogarniam takie schematy bo tak bym pomógł.....

th0m4sek, 2021-12-07:
jaki pin odpowiada za sterowanie multiplexerem i czy rezygnacja z interuptow dla MCP nie bedzie problemem przy tylu wejsciach ?

kamil2466, 2021-12-07:
Sterowanie jest zworka. Zworka założona to możliwość programowania zdjęta to wyjście na złączkę. Odnoście MCP to nie powinno być problemu, będziemy testować i zobaczymy

th0m4sek, 2021-12-07:
A może wykrywać by 5v z portu USB który podłączony będzie tylko jeśli chcemy go uzywac

th0m4sek, 2021-12-07:
A i czy gpio 39 jest do czegoś użyte?

kamil2466, 2021-12-07:
Możemy zrobić tak że jak podepnie się wtyczkę pod USB to 5v z USB przestawi na multiplexer na programowanie, a odnośnie gpio39 jest wolne bo to input a już nie mamy wolnej złączki

th0m4sek, 2021-12-07:
To podpinać 39 pod interrupts nawet jak się nie użyje to nie zaszkodzi a już 8 będzie obsługiwało przerwania ;) i będzie można potraktować ;)

kamil2466, 2021-12-07:
Ok. Jakiś pullup lub pulldown się robi ? Najlepiej przedstaw swoją propozycję na schemacie

th0m4sek, 2021-12-07:
https://arduino.stackexchange.com/questions/26898/using-interrupts-for-two-mcp23017
Tu piszą że wystarczy pullup i że nawet więcej ekspanderów można do jednego pinu spiąć.

kamil2466, 2021-12-08:
@th0m4sek napisz do mnie na PW FB lub discord

th0m4sek, 2021-12-08:
Już albo napisałem do kogoś innego na FB jak coś to na FB jestem jako Tomasz Dziliński

JakubIgnacyGorski, 2022-01-28:
Witam, zacząłem robić swoje zanim tu spojrzałem więc może podzielę się moim pomysłem na boneIO sterowane za pomocą ESP32. Liczę na poprawki mądrzejszych ode mnie gdyż jestem początkującym elektronikiem :)
[input_board_ESP32_v0.1.pdf](https://github.com/boneIO-eu/boneIO/files/7959159/input_board_ESP32_v0.1.pdf)
 
 Przy okazji chciałem się spytać tylko jak wygląda postęp Pana (@kamil2466) projektu?

kamil2466, 2022-01-28:
@JakubIgnacyGorski  nie panujemy sobie :)
Ten schemat nie aktualny, zrobony jest nowy schemat i płytka zamówiona. Powinna być na dniach u mnie. Co do twojego schematu to wszystko podobnie i tak na marginesie CP2021 ciezkie do lutowania i dlatego CH340G ja stosowałem a automat zmiany zasilania nie działa. :)

JakubIgnacyGorski, 2022-01-29:
@kamil2466 o dobrze że mówisz wywalę ten automat. I tak na razie poczekam z projektem PCB bo zaraz się okaże że niepotrzebnie to robię i twój projekt spełni moje oczekiwania ;)

maciejkrasuski, 2022-02-25:
Czy coś tu jeszcze się dzieje, czy można zamykać ten issue?

kamil2466, 2022-07-30:
tak do zamknięcia schemat nie aktualny

closed 2023-07-06 by pszafer