boneIO-eu/boneIO #91: Input ghost switching

quantum1xyz, 2022-02-25:
Samoczynne wyzwalanie inputów podczas ruchu roletami. 

BoneIO1:
Input v0.3 
Wejścia zasilane 12V
RelayBoard 32x5

BoneIO2:
Input v0.2
Wejścia zasilane 12V
ShutterBoard 32x10

Ruch rolety (230V) podłączonej do BoneIO2 wyzwala inputy w jednym i w drugim sterowniku. Nawet te, do których nic nie jest podłączone.  Niezależnie od ustawienia bounce_time. Łączenie GND zasilaczy 12V i 24V nie pomogło. Odłączenie RS485 nie pomogło. Przestawianie gpio_mode nie pomogło. Odłączenie napięcia od shutter board i pstrykanie przekaźnikami nie powoduje wyzwolenia inputów. Podłączenie żarówki w miejsce rolety i właczanie/wyłączanie nie powoduje wyzwolenia inputów.

ApacerPL, 2022-02-25:
Indukcja w uzwojeniu?

radzikr, 2022-02-25:
Czy zbadałeś które inputy się wzbudzają

quantum1xyz, 2022-02-25:
Tak, praktycznie wszystkie. Raz jedne, raz drugie. Nie ma żadnej reguły.

patryk84a, 2022-05-20:
Czy wejscia wyzwalaja sie w momdncie uruchomienia rolety, czy w czasie dzialania, czy dopiero po zatrzumaniu?

kamil2466, 2022-10-02:
Dużo siedziałem nad tematem wyzwoleń input przy działaniu rolet, niestety problemem nie są płytki input board ani relay board tylko płytka Beaglebone Black. Procesor jest bardzo wrażliwy na łuki elektryczne silników rolet ze wewnętrznie włączą gpio. Temat w BBB nie jest do rozwiązania. Testowałem to bez naszych płytek Input board na samym BBB i potwierdziłem przypuszczenia. Dla rolet zostaje rozwiązanie na ESP ale tu również trzeba pamiętać aby oddalić sterownik boneio BBB od okablowania rolet oraz sterownika boneio ESP i sterownik do rolet musi być na osobnym zasilaczu.

closed 2023-05-30 by pszafer