boneIO-eu/boneIO #102: SDM230

GieBek, 2022-09-22:
Cześć, brak obsługi licznika SDM230. 
Nie chce też się połączyć na SDM120 i 630.

pszafer, 2022-09-22:
Nie mam takiego licznika więc musi go dodać ktoś kto takowy posiada. Pozostałe SDM powinny działać.

GieBek, 2022-09-22:
Kurde, może ja mam coś źle. 
Przecież to jest dokładnie to samo co 120-stka ale na dwa pola. 

`modbus:
  uart: uart4

modbus_sensors:
  - id: sdm120
    model: sdm120
    address: 1
    update_interval: 10s`

`Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 ERROR (MainThread) [boneio.modbus] Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 WARNING (MainThread) [boneio.sensor.modbus] Can't fetch data from modbus device sdm120. Will sleep for 15.0 seconds
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 DEBUG (MainThread) [boneio.sensor.temp] Fetched temperature 27.375. Applying filters.
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 DEBUG (MainThread) [boneio.mqtt_client] Sending message topic: boneiohome/sensor/temp_boneio, payload: {"state": 27.38}`

GieBek, 2022-09-22:
Kurde, może ja mam coś źle.
Przecież to jest dokładnie to samo co 120-stka ale na dwa pola.

```
modbus:
uart: uart4

modbus_sensors:

id: sdm120
model: sdm120
address: 1
update_interval: 10s
```

```
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1 
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE' 
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 ERROR (MainThread) [boneio.modbus] Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response 
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 WARNING (MainThread) [boneio.sensor.modbus] Can't fetch data from modbus device sdm120. Will sleep for 15.0 seconds 
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 DEBUG (MainThread) [boneio.sensor.temp] Fetched temperature 27.375. Applying filters. 
Sep 22 21:15:21 beaglebone boneio[1349]: 2022-09-22 21:15:21 DEBUG (MainThread) [boneio.mqtt_client] Sending message topic: boneiohome/sensor/temp_boneio, payload: {"state": 27.38}
```

ApacerPL, 2022-09-23:
Dodaj mu w pliku w jakim trybie ma odczytywać bo z tego co widziałem to tego nie ma dla sdm120

GieBek, 2022-09-23:
O jaki tryb chodzi?
Wkleiłem wszystko z instrukcji:
[https://boneio.eu/docs/configuration/modbus-sensors](url)

ApacerPL, 2022-09-23:
Otwórz sobie plik sofara i zobacz o co chodzi, tryby holding/input. Sam błąd wskazuje, że pymodbus nie wie jak dekodować dane

GieBek, 2022-09-23:
No niestety dalej nic, w każdym razie dzięki za pomoc. 
Mam SDM630 podłączony do ESP'ka, podłączę się do niego żeby sprawdzić czy wszystko gra z licznikiem. 

Logi puki co mam teraz takie:
```
Sep 23 15:28:48 beaglebone boneio[4877]: 2022-09-23 15:28:48 DEBUG (MainThread) [pymodbus.transaction] Current transaction state - IDLE
Sep 23 15:28:48 beaglebone boneio[4877]: 2022-09-23 15:28:48 DEBUG (MainThread) [pymodbus.transaction] Running transaction 1
Sep 23 15:28:48 beaglebone boneio[4877]: 2022-09-23 15:28:48 DEBUG (MainThread) [pymodbus.transaction] SEND: 0x2 0x4 0x0 0x0 0x0 0x2 0x71 0xf8
Sep 23 15:28:48 beaglebone boneio[4877]: 2022-09-23 15:28:48 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
Sep 23 15:28:48 beaglebone boneio[4877]: 2022-09-23 15:28:48 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
Sep 23 15:29:36 beaglebone boneio[4877]: 2022-09-23 15:29:36 DEBUG (MainThread) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
Sep 23 15:29:36 beaglebone boneio[4877]: 2022-09-23 15:29:36 DEBUG (MainThread) [pymodbus.transaction] Running transaction 7
Sep 23 15:29:36 beaglebone boneio[4877]: 2022-09-23 15:29:36 DEBUG (MainThread) [pymodbus.transaction] SEND: 0x2 0x4 0x0 0x0 0x0 0x2 0x71 0xf8
Sep 23 15:29:36 beaglebone boneio[4877]: 2022-09-23 15:29:36 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Changing state to IDLE - Last Frame End - None, Current Time stamp - 1663946976.6375
Sep 23 15:29:36 beaglebone boneio[4877]: 2022-09-23 15:29:36 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
Sep 23 15:29:36 beaglebone boneio[4877]: 2022-09-23 15:29:36 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
Sep 23 15:30:10 beaglebone boneio[4877]: 2022-09-23 15:30:10 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
Sep 23 15:30:10 beaglebone boneio[4877]: 2022-09-23 15:30:10 DEBUG (MainThread) [pymodbus.transaction] RECV: 
Sep 23 15:30:10 beaglebone boneio[4877]: 2022-09-23 15:30:10 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame - [b''] not ready
Sep 23 15:30:10 beaglebone boneio[4877]: 2022-09-23 15:30:10 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
Sep 23 15:30:10 beaglebone boneio[4877]: 2022-09-23 15:30:10 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Sep 23 15:31:13 beaglebone boneio[4877]: 2022-09-23 15:31:13 DEBUG (MainThread) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
Sep 23 15:31:13 beaglebone boneio[4877]: 2022-09-23 15:31:13 DEBUG (MainThread) [pymodbus.transaction] Running transaction 19
Sep 23 15:31:13 beaglebone boneio[4877]: 2022-09-23 15:31:13 DEBUG (MainThread) [pymodbus.transaction] SEND: 0x2 0x4 0x0 0x0 0x0 0x2 0x71 0xf8
Sep 23 15:31:13 beaglebone boneio[4877]: 2022-09-23 15:31:13 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Changing state to IDLE - Last Frame End - None, Current Time stamp - 1663947073.5262
Sep 23 15:31:13 beaglebone boneio[4877]: 2022-09-23 15:31:13 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
Sep 23 15:31:13 beaglebone boneio[4877]: 2022-09-23 15:31:13 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
Sep 23 15:31:46 beaglebone boneio[4877]: 2022-09-23 15:31:46 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
Sep 23 15:31:46 beaglebone boneio[4877]: 2022-09-23 15:31:46 DEBUG (MainThread) [pymodbus.transaction] RECV: 
Sep 23 15:31:46 beaglebone boneio[4877]: 2022-09-23 15:31:46 DEBUG (MainThread) [pymodbus.framer.rtu_framer] Frame - [b''] not ready
Sep 23 15:31:46 beaglebone boneio[4877]: 2022-09-23 15:31:46 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 2
Sep 23 15:31:46 beaglebone boneio[4877]: 2022-09-23 15:31:46 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Sep 23 15:31:46 beaglebone boneio[4877]: 2022-09-23 15:31:46 ERROR (MainThread) [boneio.modbus] Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response
Sep 23 15:31:46 beaglebone boneio[4877]: 2022-09-23 15:31:46 WARNING (MainThread) [boneio.sensor.modbus] Can't fetch data from modbus device SDM120. Will sleep for 15.0 seconds
Sep 23 15:31:46 beaglebone boneio[4877]: 2022-09-23 15:31:46 DEBUG (MainThread) [boneio.sensor.temp] Fetched temperature 27.25. Applying filters.
```

closed 2023-05-30 by pszafer