boneIO-eu/boneIO #105: Analog input critical bug

krystiansierota, 2022-11-07:
W dokumentacji jest napisane że obsługuje do 9.9V.
1. Nie obsługuje takiego napięcia wejsciowego bo ten wzmacniacz co robi za bufor jest zasilany z 5V a sygnał z zacisku wchodzi bezposrednio na niego więc na wyjściu ze wzmacniacza nie będzie więcej niż 5V a nawet i tego nie będzie bo to nie jest wzmacniacz rail to rail. W zasadzie zgodnie z dokumentacją to nie można na jego wejscie dać więcej niż Vcc-1.5V czyli 3.5V.
2. Takie nieobciążone wejście to bedzie jeden wielki szum a nie pomiar. Wejście analogowe trzeba obciążyć do masy rezystorem np. 47k (proszę popatrzeć jakie są impedancje wejściowe np sterowników PLC na wejściach analogowych)
3. Bezpośrednie podłączenie wejścia wzmacniacza do szyny bez żadnego zabezpieczenia ESD/overvoltage to w sumie jednorazówka. Do pierwszego wyładowania od palucha i po wzmacniaczu.

Na wejściu wzmacniacza oprócz zabezpieczeń i obciążenia to by się jeszcze jakiś filtr RC przydał bo bez niego to te pomiary ADC to też za wiele sensu nie będą miały

kamil2466, 2022-11-08:
Dziękuje za zgłoszenie błędu.
Wersja uległa modyfikacją i poprawką nie jest jeszcze udostępniona ponieważ jeszcze ją sprawdzamy. Co do ADC chcieliśmy przejść na konwerter analogowo cyfrowy po magistrali i2c [ADS1115].
Jeżeli chciałbyś pomóc udoskonalić projekt to zapraszam.

kamil2466, 2022-11-10:
Jaką masz propozycję naprawy ?
Po ustaleniach chcemy wprowadzić dzielnik napięć od 0-25v 
[image]
[image]

Jeżeli macie inne pomysły to proszę o przykłady poniżej

krystiansierota, 2022-11-10:
W obecnej sytuacji beż zmiany wzmacniacza ten dzielnik jest raczej jedyną
opcją

donpablos, 2022-11-12:
Mam już jedno beaglebone spalone, gdyż chciałem wykryć napięcie na ADC.

  Mam zasilacz na listwie din 5V DC ( sygnalizuje ze poziom szamba jest przekroczony) i najbardziej mi pasuje abym mógł wykrywać prąc 5V DC. Jaki jest sposób z obecnym boneio aby wykryć te 5V? Czy wystarczy dodać rezystor(jaki??) przed ADC ? Czy trzeba zrobić coś jeszcze( drugi rezystor gdzieś na płytce przy GND??) ?

pszafer, 2022-11-12:
Wątek dotyczy boneIO ESP. Stwórz własny lub zapraszamy na discord

ApacerPL, 2022-11-20:
Uklady ads7830 może użyć do pomiaru?