boneIO-eu/boneIO #110: Add contact info / discussion board

blaa, 2023-02-15:
Hi,
  I was looking for a way to get in touch - some discussion board, irc, matrix, discord, list of emails, whatever. But couldn't find any link on github or in docs. So that's what this "issue" is about: add some to docs. ;) With a language preference. If it's there and I missed it - sorry, I tried.

On the seemingly unrelated note I'm currently finishing a house, have cables prepared and I was going to build my own smart home implementation, but maybe I can reuse and contribute somehow to boneIO.

My idea so far was to use STM32 platform with a CAN bus as a main communication interface (mostly star-shaped) and occasional interfacing with modbus/RS485 - depending on interfaces I can buy or find in appliances. Code µC parts in Rust for additional stability/robustness and some challenge. Tie everything with homeassistant (maybe with a USB ELM327, or own bridge).

But I considered going with ESP32 and Micropython too. I know Python/C did some STM32, AVR, ESP32 programming in the past and host homeassistant.

So if you have some communication medium I could linger around.

pozdrawiam,

sokrates209, 2023-02-15:
@blaa please join Discord :). Links are on the website:
https://boneio.eu/

blaa, 2023-02-15:
Ah, yes in footer. I though I will just miss it. Could be added to the README on gh though. ;) Consider.

closed 2023-02-15 by blaa